KUL-recon-lab/parallelproj #2: create CMake export files for use of parallelproj by external projects

KrisThielemans, 2021-02-10:
See https://cmake.org/Wiki/CMake/Tutorials/Exporting_and_Importing_Targets

relevant STIR code is at https://github.com/UCL/STIR/blob/master/CMakeLists.txt#L215-L244 which could serve as an example

gschramm, 2021-02-10:
export and install of target is now also added in CMakeLists.txt. The cmake files are installed in the same dir as the libs (not sure if this is optimal).

KrisThielemans, 2021-02-10:
there's a bit more to it I'm afraid. It needs the `CONFIGURE_PACKAGE_CONFIG_FILE` lines etc. The point is that it should create a `xxxConfig.cmake` (and other files) which `find_package` will go hunting for. (In my original comment I also forgot to link to the [version lines](https://github.com/UCL/STIR/blob/a08cb7504aac957742e2b7a81240a38a62f2e489/CMakeLists.txt#L182-L186)).

Also, you shouldn't really install includes, libs  and cmake configs in `CMAKE_CURRENT_LIST_DIR`, but use `CMAKE_INSTALL_PREFIX`. Traditional location for the cmake configs seems to be `${CMAKE_INSTALL_PREFIX}/lib/cmake`

By the way, I personally find the name `pyparallelproj` a bit confusing for a project called `parallelproj` (and we won't use the Pyhton part.

gschramm, 2021-02-11:
(1) so having a structure like (assuming that CMAKE_INSTALL_PREFIX=/usr/local/lib) would work for you:
/usr/local/lib/parallelproj/libparallelproj.so
/usr/local/lib/cmake/parallelproj-targets.cmake
/usr/local/lib/cmake/parallelproj-targets-release.cmake
/usr/local/lib/cmake/....

(2) do you know an elegant way to include an automatic version number (e.g. derived from git describe --tags)
     in cmake?

gschramm, 2021-02-13:
(1) I worked on parallelprojConfig.cmake.in, targets and version such that

parallelprojConfig.cmake
parallelproj-targets.cmake
parallelproj-targets-release.cmake
parallelprojVersion.cmake

Are now generated and installed.
**A zip containing all files installed in an example build is attached**
[parallelproj_install.zip](https://github.com/gschramm/parallelproj/files/5975636/parallelproj_install.zip)

In this build, CUDA was available so the standard c-only openmp lib and the
optional CUDA lib were built.

Can you have a look if that is sufficient for the STIR build process?

(2) One thing that is still unclear to me is whether I need to specify
PATH_VARS in onfigure_package_config_file()
https://github.com/gschramm/parallelproj/blob/master/CMakeLists.txt#L84

KrisThielemans, 2021-02-16:
All works for me.

closed 2021-02-16 by KrisThielemans