KUL-recon-lab/parallelproj #5: correct behavior of fwd and back projectors is start and end points are within the image cube

gschramm, 2021-02-17:
right now, the fwd and back projectors step through all planes which means that voxels in the image cube that are "outside" on the LOR also contribute. This should be fixed by adapting the ray_cube_intersection functions.

gschramm, 2021-02-18:
@KrisThielemans behavior of the fwd and back projector was changed such that only voxels between the LOR start and end point contribute. Can you check if that is similar to the behavior of the STIR projectors?

KrisThielemans, 2021-02-18:
I'm not so sure if we are talking about the same thing. I wasn't talking about the fact that the STIR backprojection was circular and yours not.
In the image in the PR, I ran the STIR ray-tracing projector with a circular FOV (size determined by the image). It can also use the whole image.  In many cases that is just a waste of time as there is no activity there of course, but it's up to the user.

You can choose to do either (or give the option), I don't mind.

However, I was referring to the "lobes" at the outside circle. I'm not sure where they come from.

Note that in STIR there is currently no check that the image is actually smaller than the ring diameter, so we could have the situation where the ray-tracing goes too far. But that wasn't the case when I ran it.

gschramm, 2021-02-18:
do you mean the white spots? in the latest commit the the back projector uses all the voxels between the LORs, so I would predict weird behavior in the voxels containing detectors. limiting the transaxial FOV would be an option that I write on the TODO list. But usually for "small" (brain) FOV I cover this by passing a smaller volume (which is of course not exactly the same)

KrisThielemans, 2021-02-18:
yes, the white features in the corners
[image]

I'm not sure that it's the detectors. The corner coordinates of the image are at "x,y" 296.25,296.25mm.  The inner ring diameter of the mMR is 656mm. But maybe I gave it the wrong coordinates of course. I'll have to check that later.

gschramm, 2021-02-19:
@KrisThielemans I think those are the gaps between the detector module. At least this is what I see when I simulate modulized polygon scanner and choose are volume where the corners are outside the ring. 

In the original issue you mentioned that the data is from the SIGNA (not the mMR). The signa has transaxial 28 modules, so you should see 28 white blobs.

gschramm, 2024-12-14:
@KrisThielemans did we solve this issue?

KrisThielemans, 2024-12-17:
I don't know for sure, and won't be able to test for a while. sorry. Feel free to close.

closed 2024-12-17 by gschramm