KUL-recon-lab/parallelproj #7: document that the forward projector accumulates

KrisThielemans, 2021-02-19:
Admittedly, I didn't read the documentation... But it surprised me that forward projection adds to the existing data (but it's fine, as long as we know)

gschramm, 2021-02-22:
Well spotted, this was actually a point where the C/OpenMP and CUDA implementation behaved differently  - which was corrected.
Now in both implementations the projection arrays are initialized with 0s before the fwd projection is done.
Initializing with 0s makes more sense, since in CUDA it avoids unnecessary memory transfer, and the overhead in the
OpenMP implementation should be negligible.

The documentation is also updated (mentioning that in the back projection values are added to the existing array).

closed 2021-02-22 by gschramm

KrisThielemans, 2021-02-22:
ok. I guess we could skip the fill() in STIR then...

gschramm, 2021-02-22:
definitely. sorry for the inconvenience.