KUL-recon-lab/parallelproj #9: Installing parallelproj for STIR errors

robbietuk, 2021-03-10:
Hi,

This is related to https://github.com/UCL/STIR/issues/840. 

I am trying to install parallelproj for use in the new STIR GPU projector on a cluster system (note I do not have admin privilages). 

Copy/Paste:
-----
I have installed parallelproj using the method detailed [here](https://github.com/gschramm/parallelproj/blob/master/README.md#i-compilation-of-openmp-and-cuda-libraries), by calling:
```
python build_libs_for_python.py
```

I set `parallelproj_DIR=/wherever/lib_Linux_64bit/cmake` (it seems it creates `lib_Linux_64bit` instead of `lib`) and add it to my `$PATH`. The cmake configuration reports:
```
 Found parallelproj (will use its CUDA support)
 Parallelproj projector support enabled.
```

STIR installs as normal but then I get the following errors when calling many STIR executables.
```
[******@login13 build]$ forward_project 
forward_project: error while loading shared libraries: libparallelproj_c.so: cannot open shared object file: No such file or directory
```

_________

Should I be attempting install with python or should I do my own custom cmake install? Note, I do not have admin privilage so I need to install everything to a custom location.

gschramm, 2021-03-10:
Hi Robbie,

If you want to use parallelproj for STIR, don't use my python build script. The intention of that script is just to trigger a build that installs the libs at a specific location such that my python wrappers know whey they are (inside a python package).

For use with STIR, I would do a "normal" STIR build and setting the CMAKE_INSTALL_PREFIX to a path where you have 
enough privileges. 

@KrisThielemans can maybe help with how to tell the STIR build that parallelproj was build and installed at your specific location.

KrisThielemans, 2021-03-10:
standard CMake stuff I'm afraid
```
cmake -Dparallelproj_DIR=/blab/bla/lib/cmake STIR-source`
```
(probably not necessary to specify the `lib/cmake` part). The wasn't necessary in the SuperBuild](https://github.com/SyneRBI/SIRF-SuperBuild/blob/master/SuperBuild/External_STIR.cmake), as CMake checks locations specified by `INSTALL_PREFIX` etc etc. See https://cmake.org/cmake/help/latest/command/find_package.html#search-procedure

robbietuk, 2021-03-10:
Using `cmake -B . -DCMAKE_INSTALL_PREFIX=/home/robbie/bin/parallelproj_bin/test/install ../parallelproj` correctly sets the `CMAKE_INSTALL_PREFIX` but the `PARALLELPROJ_INSTALL_INCLUDEDIR`, `PARALLELPROJ_INSTALL_LIBDIR`, and `PARALLELPROJ_INSTALL_DOCDIR` all remain default. By manually setting these to be within the `CMAKE_INSTALL_PREFIX` directory, I was able to build and use STIR executables again.

KrisThielemans, 2021-03-10:
was this on a clean build? Otherwise, these are likely cached variables, so what remain what they were.

robbietuk, 2021-03-10:
Sorry, my cmake is not very good. Is this something related to https://github.com/gschramm/parallelproj/blob/7124cc56014e842e3884a7d8fe341364861a73ae/CMakeLists.txt#L23-L32

gschramm, 2021-03-10:
If I run the following in a build subdir
```
cmake -DCMAKE_INSTALL_PREFIX=/foo/bar ..
make
make install
```
everything (lib,include,share/doc) gets nicely installed in "/foo/bar" as expected

KrisThielemans, 2021-03-10:
it does so for the superbuild  as well. Robbie, are you sure you didn't run cmake first without `CMAKE_INSTALL_PREFIX`?  One way where this often happens is if you run it from the GUI. You press Configure, CMake will run through the whole thing, set the cached variables, and it's only afterwards that you can (easily) modify the `CMAKE_INSTALL_PREFIX`, by which time it is too late. (We have the same problem in STIR/SIRF CMake with `PYTHON_DEST` etc).

If we don't need the `PARALLELPROJ_INSTALL*`  variables cached (i.e. user-settable), then that problem will disappear

robbietuk, 2021-03-10:
Okay fixed, thank you!
```
cmake -DCMAKE_INSTALL_PREFIX=/foo/bar ..
make
make install
```

This installed `lib64` and `include` into the `DCMAKE_INSTALL_PREFIX` directory. I think the issue was a part of my debugging step, opening the gui `ccmake .` in the build directory, i.e. 
```
cmake -DCMAKE_INSTALL_PREFIX=/foo/bar ..
ccmake . (check variables)
make
make install
```

closed 2021-03-10 by robbietuk