KUL-recon-lab/parallelproj #22: build failure without CUDA

KrisThielemans, 2022-12-29:
I tried to build v1.2.9, admittedly on Ubuntu 16.04 VM, but I don't think that matters, and get
```
Compiler requires the CUDA toolkit.  Please set the CUDAToolkit_ROOT
  variable.
```
It looks like I need to set `SKIP_CUDA_LIB`. That's ok  I guess, but it creates some difficulty for the SIRF-SuperBuild. We'll have to detect CUDA first and then set the variable accordingly. Not too hard I suppose (we probably do this already a bit).

It's not so easy to decide what the best way forward for this. @gschramm what do you think?

gschramm, 2022-12-30:
I think detecting whether CUDA is present upfront and setting ```-DSKIP_CUDA_LIB``` and ```-DCMAKE_CUDA_ARCHITECTURES=all``` otherwise, makes sense.
In the conda-forge recipies, [I do those decisions that way](https://github.com/conda-forge/parallelproj-feedstock/blob/1485d5dd5642b98dffe23798ad3c986860d7bbcc/recipe/build.sh#L7). But that only works, because when using
the conda-forge actions, I can test for ```${cuda_compiler_version:-None} != "None"```

KrisThielemans, 2023-01-06:
> For me, removing all the cmake CUDA detection magic and purely relying on the "skip" flags, seems more robust.

This is true when building `parallelproj` by hand. But in an automated system where the build should work anywhere, it is difficult. Anyway, we'll see what we can do in the SIRF-SuperBuild, which is currently failing https://github.com/SyneRBI/SIRF-SuperBuild/pull/787#issuecomment-1373307637

closed 2023-03-01 by gschramm