KUL-recon-lab/parallelproj #23: linking error with clang

KrisThielemans, 2022-12-29:
We have a (non-CUDA) Action compiling with `clang` but get
https://github.com/UCL/STIR/actions/runs/3803702720/jobs/6470298010#step:4:405
```
[100%] Linking C executable parallelproj_nontof_fwd_back_test
/usr/bin/ld: ../libparallelproj_c.so.1.2.9: undefined reference to `floorf'
/usr/bin/ld: ../libparallelproj_c.so.1.2.9: undefined reference to `sqrtf'
/usr/bin/ld: ../libparallelproj_c.so.1.2.9: undefined reference to `ceilf'
```

KrisThielemans, 2022-12-29:
testing with v1.2.9 by the way, in case that matters

gschramm, 2022-12-30:
Hi Kris,
I think this is because my [check for whether I need to link against libm](https://github.com/gschramm/parallelproj/blob/1822f9bfc53f8bf7f37e5eca1f071a16608a8f05/c/tests/CMakeLists.txt#L17) is wrong (only works correctly for gcc and msvc).
I guess the line should be:
```
if(NOT MSVC)
```
Do you know a better / more robust way to test whether I need to link against libm?

KrisThielemans, 2022-12-30:
https://stackoverflow.com/a/63037953/15030207 suggests to check if the library exists, and then link, which seems sensible to me, so why not
```cmake
if(LIBM)
    message(STATUS "linking test against libm")
    list(APPEND POW_LIBS "m")
    target_link_libraries(parallelproj_nontof_fwd_back_test PUBLIC ${POW_LIBS})
endif()
```
or even
```
if(LIBM)
    message(STATUS "linking test against libm")
    list(APPEND POW_LIBS "m")
endif()
target_link_libraries(parallelproj_nontof_fwd_back_test PUBLIC ${POW_LIBS})
```
which might work even if `POW_LIBS` is empty.

Weirdly enough, we don't seem to need this in STIR (or at least, I cannot find it nor remember it).

gschramm, 2022-12-30:
(1) If I compile locally with clang on MacOS, I also don't need to link against libm :/

(2) I guess this approach from [here](https://cliutils.gitlab.io/modern-cmake/chapters/features/small.html) is even better:
```
find_library(MATH_LIBRARY m)
if(MATH_LIBRARY)
    target_link_libraries(MyTarget PUBLIC ${MATH_LIBRARY})
endif()
```

KrisThielemans, 2022-12-30:
simpler for sure.

gschramm, 2022-12-30:
One more question due to my lack of C linking knowledge:
Since I am using e.g. ```floorf``` and ```sqrtf``` in the OpenMP projection libs,
do I need to link against ```libm``` when I compile the shared libs? 
Or, do we only link against ```libm``` when compiling a target that uses the projection lib?

If I look at ```libparallelproj.so``` as a dependency of a target, it would be nice if by linking against ```libparallelproj.so``` all its "dependencies" would be resolved automatically.

KrisThielemans, 2022-12-30:
good point, you probably should do
```cmake
target_link_lbraries(parallelproj PUBLIC ${MATH_LIBRARY})
```
(possibly `PUBLIC` could be `INTERFACE` or even `PRIVATE`. I'm struggling a bit with those, but in this case `PUBLIC` wouldn't hurt)

gschramm, 2022-12-30:
I just pushed the change:
```
target_link_lbraries(parallelproj_c PUBLIC ${MATH_LIBRARY})
```
such that I don't need to explicitly link against ```libm``` in the test binary anymore.
the github actions to compile + test on windows with msvc and ubuntu with gcc were successful.

@KrisThielemans before creating a new release, is there a way you can test whether this solves the issue above?
I could e.g. create a temporary tag for the current parallelproj version if that would help.

KrisThielemans, 2022-12-30:
yes, 2ecde1602f76a4595d3ea9970b0d729185786c92 worked, see https://github.com/UCL/STIR/actions/runs/3808118646/jobs/6478429376#step:4:385. Thanks!

I could let STIR GHA checkout this hash, or do you intend to tag soon?

gschramm, 2022-12-30:
[tag v1.2.10 is out](https://github.com/gschramm/parallelproj/releases/tag/v1.2.10)

I will update the conda-forge feedstock tomorrow

closed 2022-12-30 by KrisThielemans