KUL-recon-lab/parallelproj #48: error when import parallelproj

Seanlam1124, 2024-01-26:
i'm running parallelproj on m1 mac with python 3.8. the error message. 
it showed up when I update parallelproj to version 1.7.1. is it because of my python version or os system

The error message:
``` bash
Traceback (most recent call last):
  File "/opt/anaconda3/envs/py4work/lib/python3.8/site-packages/IPython/core/interactiveshell.py", line 3508, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-9-1e3692be8fc6>", line 20, in <module>
    import parallelproj
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "/opt/anaconda3/envs/py4work/lib/python3.8/site-packages/parallelproj/__init__.py", line 1, in <module>
    from .backend import (
  File "/Applications/PyCharm.app/Contents/plugins/python/helpers/pydev/_pydev_bundle/pydev_import_hook.py", line 21, in do_import
    module = self._system_import(name, *args, **kwargs)
  File "/opt/anaconda3/envs/py4work/lib/python3.8/site-packages/parallelproj/backend.py", line 25, in <module>
    cuda_present = distutils.spawn.find_executable("nvidia-smi") is not None
AttributeError: module 'distutils' has no attribute 'spawn'
```

gschramm, 2024-01-26:
Many thanks for reporting this!
I just created a new test env. on my M1 Mac, and I don't get the spawn error (but a different one).
It seems indeed that I need to require python 3.9.

To test this, could you:
Create a fresh env with python 3.9 and parallelproj and try again?

e.g. using mamba
```
mamba create -n pp-test python=3.9 parallelproj
```
or 
```
conda create -c conda-forge -n pp-test python=3.9 parallelproj
```

Seanlam1124, 2024-01-26:
it did work when change to 3.9. no more error shown.
but I also tried on linux, which is also with python 3.8, it import parallelproj without any error (both specific release is 3.8.17). 
anyway, thanks for you instantly help! parallelproj helps me a lot. thanks for your hard work.

gschramm, 2024-01-26:
Always welcome. I already submitted a PR to conda-forge to force python>=3.9 (new build should appear in ca 30min).
Version 1.7.2 is also on the way that will require python 3.9 and also replace `distutils.spawn` by `shutil.which`.
(python 3.9 is required for another reason as well ;)

Many thanks for finding + reportin this!

Seanlam1124, 2024-01-26:
Hi there, i'm using parallelproj to set up my geometry. visualization is a very good update, but i wonder is there a function for setting up geometry just by inputting the crystals coordinate in 3D, which could easily access from castor. or can i set the lor position, like ring position, instead of lor spacing. because i've met a 2 different spacing in LOR spacing like this.
[image]

if I can set up geometry by crystals coordinate, it can also avoid the problem of the order of crystals, since in current result, the crystal order starts from the top panel, but mine starts from the right side.

gschramm, 2024-01-26:
1. According to your sketch, you should be able to solve that with stack of polygons, shouldn't you? It seems that you have a polygon where every side has 2 crystals (end points) and the ring spacing is non-uniform (which is supported).

2. Creating custom "ring-like" modules is also possible, but you have to similar to:
https://github.com/gschramm/parallelproj/blob/b326d4480abb2b0d0447ee48cdc1e1352dc43438/python/parallelproj/pet_scanners.py#L191

The important thing is to subclass `PETScannerModule`. When doing so, you can define any custom "ring-like" module.

But I have to adapt the input arguments to https://github.com/gschramm/parallelproj/blob/b326d4480abb2b0d0447ee48cdc1e1352dc43438/python/parallelproj/pet_scanners.py#L191 to allow to directly passing 
a module + ring positions.

closed 2024-01-26 by gschramm