KUL-recon-lab/parallelproj #60: Not reasonable result

LexieSunn, 2024-02-09:
Hi, I have a follow up question.

Currently I have 3D PET volume with size (344, 127, 344) from brainWeb and want get sinograms from it, do you have recommend settings for scanner and LOR descriptor? I tried some but not look reasonable. Code here:

`num_rings = 5
scanner = parallelproj.RegularPolygonPETScannerGeometry(
     xp,
     dev,
     radius=400.0,
     num_sides=32,
     num_lor_endpoints_per_side=16,
     lor_spacing=4.3,
     ring_positions=xp.linspace(-70, 70, 36),
     symmetry_axis=1,
 )

lor_desc = parallelproj.RegularPolygonPETLORDescriptor(
    scanner,
    radial_trim=10,
    max_ring_difference=2,
    sinogram_order=parallelproj.SinogramSpatialAxisOrder.RVP,
)

proj = parallelproj.RegularPolygonPETProjector(
    lor_desc, img_shape=(344, 127, 344), voxel_size=(2.0, 2.0, 2.0)
)`

And get this result:

[image]

Thanks for your time!

gschramm, 2024-02-09:
@LexieSunn that all depends on what you want to investigate.
I suggest to : 
- first think about what kind of scanner you want to investigate (whole-body, brain, small animal ...)
  and define a scanner matching this (based on the pre-defined ones)
- if you want to investigate effect due to limited resolution, you also have to think about how to model
  those. e.g. via (position-dependend) blurring in sinogram.

LexieSunn, 2024-02-09:
Thank you so much!

closed 2024-02-09 by LexieSunn