KUL-recon-lab/parallelproj #79: Questions regarding the positive and negative signs of the TOF-bin numbering

zjutk, 2024-07-05:
Dear Professor Georg Schramm,

I have some questions regarding the PET TOF listmode projector and the simulation of listmode events with TOF in the convert_sinogram_to_listmode function. I noticed that the TOF-bins for these listmode events are arranged as [-k, ..., 0, ..., k]. Could you please clarify the convention for the positive and negative signs of the TOF-bin numbering? Specifically, does TOF-bin[-k] represent the bin closer to one end of the LOR? For instance, does TOF-bin[-k] indicate the bin is closer to the xstart end, and TOF-bin[k] closer to the xend end?

Thank you for your assistance.

gschramm, 2024-07-05:
Dear Kun,

1. excellent question. The TOF-bin[k] is closer to `xstart`, TOF-bin[0] is in the center between `xstart` and `xend` and, TOF-bin[k] is closer to `xend`.  See e.g. these lines mentioned below. `u` is the normalized directional vector pointing from `xstart` to `xend`.

2. As always in programming, writing a simple unit test to test your hypothesis is good practice. You can e.g. back project a TOF listmode events on the same LOR defined by `xstart` and `xend` varying the TOF bin and see where the how the back projection changes. 

Bests,
Georg

https://github.com/gschramm/parallelproj/blob/1f83557cdc06ff3368448f7d301ad1ea6271fe94/c/src/joseph3d_fwd_tof_lm.c#L103
https://github.com/gschramm/parallelproj/blob/1f83557cdc06ff3368448f7d301ad1ea6271fe94/c/src/joseph3d_fwd_tof_lm.c#L144
https://github.com/gschramm/parallelproj/blob/1f83557cdc06ff3368448f7d301ad1ea6271fe94/c/src/joseph3d_fwd_tof_lm.c#L195

zjutk, 2024-07-05:
Dear Professor Georg Schramm,

Thank you very much for your response! Regarding your first point, my current understanding is that TOF-bin[k] is closer to xstart, while TOF-bin[-k] is closer to xend. Could you please confirm if my understanding is correct?

Additionally, I have another question. Suppose there are two events with xstart and xend in opposite directions, for example:

start_ring = xp.asarray([2,1])
start_xtal = xp.asarray([0,90])

end_ring = xp.asarray([1,2])
end_xtal = xp.asarray([90,0])

event_tof_bins = xp.asarray([5, 5])

When using the convert_sinogram_to_listmode function for simulation, will such events be merged into a single representation with the same xstart and xend, and with opposite signs for event_tof_bins? For example:

start_ring = xp.asarray([2,2])
start_xtal = xp.asarray([0,0])

end_ring = xp.asarray([1,1])
end_xtal = xp.asarray([90,90])

event_tof_bins = xp.asarray([5, -5])

In other words, does the convert_sinogram_to_listmode function produce listmode data that only contains one direction from xstart to xend?

Thank you once again for your helpful assistance!

Best regards,
Kun

gschramm, 2024-07-05:
> Dear Professor Georg Schramm,
> 
> Thank you very much for your response! Regarding your first point, my current understanding is that TOF-bin[k] is closer to xstart, while TOF-bin[-k] is closer to xend. Could you please confirm if my understanding is correct?

No it is the other way around, as I wrote in my last comment.

> Additionally, I have another question. Suppose there are two events with xstart and xend in opposite directions, for example:
>
> start_ring = xp.asarray([2,1]) start_xtal = xp.asarray([0,90]) [...]

Sinogram to listmode converts (as the name says) an (integer emission) sinogram to a listmode file. 
But you are describing the reverse operation (binning LM events into a sinogram).

> 
> Thank you once again for your helpful assistance!
> 
> Best regards, Kun

zjutk, 2024-07-08:
Dear Professor Georg Schramm,

Thank you very much for your explanation! Regarding my first point of confusion, in your last comment you wrote, “...The TOF-bin[k] is closer to xstart,..., TOF-bin[k] is closer to xend.” I am still a bit unclear about this statement. The two TOF-bin[k]s you mentioned—are they referring to the same TOF-bin[k], or are they TOF-bin[-k] and TOF-bin[k] located at the two ends of the LOR, respectively?

By the way, I have another question I would like to ask. In your previous response to my query in the pyparallelproj project, you mentioned that “start” and “end” are not related to the detection time point of the photons. Does this still apply in the current parallelproj framework, especially for listmode TOF events?

Best regards,
Kun

gschramm, 2024-07-08:
Hi Kun,

1. TOF-bin[-k] is closer to xstart, TOF-bin[+k] is closer to xend
2. Indeed, the definition of xstart and xend is purely geometric and not related TOF differences

zjutk, 2024-07-08:
Dear Professor Georg Schramm,

Thank you very much for your reply! It perfectly resolved the issue I was experiencing.
Best regards,
Kun

closed 2024-07-08 by gschramm