KUL-recon-lab/parallelproj #81: Issue regarding the negative Poisson log-likelihood function using sinogram data and list-mode data

zjutk, 2024-07-19:
Dear Professor Georg Schramm,

Thank you very much for your previous response regarding my question about the log-likelihood function! Based on my understanding of your reply, the Ote version of the list-mode negative Poisson log-likelihood function can be rewritten as:

$logL = \sum_i \bar{y_i}(x) - y_i \log( \bar{y_i}(x)) $

However, I noticed that this form is consistent with the negative Poisson log-likelihood function using sinogram data mentioned in this example(https://parallelproj.readthedocs.io/en/stable/auto_examples/05_algorithms/01_run_mlem_basic.html#sphx-glr-auto-examples-05-algorithms-01-run-mlem-basic-py), but differs from the list-mode negative Poisson log-likelihood function described here(https://parallelproj.readthedocs.io/en/stable/auto_examples/06_listmode_algorithms/01_listmode_mlem.html#sphx-glr-auto-examples-06-listmode-algorithms-01-listmode-mlem-py):

$logL = \sum_i \bar{y}_i (x) - \bar{y}_i (x) \log(y_i)$

Could you please clarify whether these two formulas are equivalent? If not, which one is the correct list-mode negative Poisson log-likelihood function?

Thank you again for your assistance.

Sincerely,
Kun Tian

gschramm, 2024-07-19:
Sorry for the confusion, what is called "logL" above is the negative Poisson log likelihood.
We sometimes use that because in optmization, many people prefer to minimize vs maximize. 

$$-logL = \sum_i \bar{y_i} - y_i \log \bar{y_i} $$

The Poisson logL is indeed (easy to check yourself):

$$ logL = \sum_i y_i \log \bar{y_i} -  \bar{y_i} $$

closed 2024-07-19 by gschramm

zjutk, 2024-07-19:
Thanks for your reply! However, my main question is: in the list-mode negative Poisson log-likelihood function, should it be $\(\bar{y}_i (x) \log(y_i)\)$ or $\(y_i \log \bar{y}_i(x)\)$?

gschramm, 2024-07-19:
the second 

$$ logL y_i \log \bar{y_i} $$

easy to see if you know the Poisson distribution:

$$ \frac{\exp(-\bar{y_i}) \bar{y_i}^{y_i}}{y_i !}   $$