KUL-recon-lab/parallelproj #88: `cp` not found

lukepolson, 2024-10-01:
If `torch` is installed but `cupy` is not installed, then `is_cuda_array(img)` will trigger true in line 482 of `src/parallelproj/backend.py` but then it will fail since `cp` is not imported. This pull request fixes a few of these bugs and forces run on CPU if CuPy is not installed.

gschramm, 2024-10-02:
Hi Luke, thanks for reporting this issue.

1. Can you confirm that `is_cuda_array(torch.ones((2,2,2), device = "cpu") = True`. It is supposed to return `False`.
The idea of `is_cuda_array` is to return `True` for cupy arrays and torch gpu tensors and `False` otherwise.

2. I agree that if people want to use torch with gpu tensors, than cupy should be installed as well. But that should be fixed in the installation instructions.

Georg

lukepolson, 2024-10-02:
The issue is that you can have `torch` tensors on GPU even if `cupy` is not installed. In that case, all the `if` statements return true, but `cp` is never defined because line 29 yields `cupy_enabled` as False.

lukepolson, 2024-10-02:
Ah I see what you're saying. The pull request should probably go along with the change in installation instructions, just in case somebody has PyTorch but not Cupy. In the present case, they'll get a `cp not defined` error, but after the pull request, it will simply run on cpu. Maybe I can add some sort of warning in the pull request if you think that would be best, e.g. "Warning, CuPy install needed for GPU implementation, currently running on CPU", or something like that

gschramm, 2024-10-02:
There are many scenarios where people might run pytorch without cupy (e.g. me on my macbook using pytorch CPU tensors).
So using pytorch without cupy is fine - just not in the case when you want to use pytorch CUDA tensors.

I will update the install instructions stating that if people want to use pytorch CUDA tensors, cupy needs to be there as well.

lukepolson, 2024-10-02:
Ah apologies I mis-spoke on the last comment, what I meant is if somebody has pytorch-GPU installed without CuPy: in this case the library breaks. But yes, I see what you're saying!

gschramm, 2024-10-02:
Exactly. So we need to make sure that the install instructiongs clearly say that if you want torch CUDA, you also need cupy.

lukepolson, 2024-10-02:
Sure, but I would also suggest updating the code with #87 in case somebody installs without installing cupy. And maybe add some sort of warning as well. (A scenario I can think of is that somebody installs `parallelproj`, then later installs `torch` for some machine learning or something, and then finds that the code stops running).

lukepolson, 2024-10-02:
Hi Georg, I agree that updating the install instruction is important. But keep in mind that the environment can change post install, and this might cause the code to break.

For (2), it looks like CuPy (for example) [builds various wheels for different cuda versions](https://github.com/cupy/cupy/releases/tag/v13.3.0), as you mention what is done on conda. They do this through a separate library for building called [cuda-release-tools](https://github.com/cupy/cupy-release-tools/tree/main). After a brief look, it seems like they manually set up different OS/CUDA configurations and build each wheel that way. Definitely not a simple solution, especially given (as you said) that the conda-forge people have solved this.

gschramm, 2024-10-02:
"environment can change post install", true but people need to track their depedencies and afaik, any decent package manager won't delete packages. So installing toch and cupy together with parallelproj from the start is the key.

Note that package resolution can be difficult which is why changing the env. sometimes is tricky / creates conflicts.

gschramm, 2024-10-03:
@lukepolson : see #89 and https://parallelproj--89.org.readthedocs.build/en/89/installation.html (new read the docs build based on PR). What do you think about the install instructions now?

lukepolson, 2024-10-03:
Hi Georg, it seems much better now as it enforces `cupy` to be installed. Thanks :)

closed 2024-10-04 by gschramm