KUL-recon-lab/parallelproj #90: Problem with FBP using projector including resolution model and the effect of attenuation

hamidrkKTH, 2024-10-22:
Hi George,

Hope you are doing well.
I am using parallelproj in my experiments for PET image reconstruction. In one part, I create a small projector that includes resolution model and the effect of attenuation. Then I use this projector to create sinogram from the input PET data and then try to perform back projection and Filter Back Projection (FBP).
However, when I compare the results (attached), the FBP of the sinogram generated using the complete projector (with both resolution model and attenuation) appears worse than the FBP of the sinogram created using the projector with only the resolution model. (the image on the bottom right is worse than the one to its left.).
I followed the provided example codes exactly. Could you please help me identify where the issue might be? Here is the code and the results:

x_fwd1 = proj(f_gpu)  # "f_gpu" is the input PET image
 # Incorporating image-based resolution model
res_model = parallelproj.GaussianFilterOperator(
    proj.in_shape, sigma=0.75 / (2.35 * proj.voxel_size)
)
proj_with_res_model = parallelproj.CompositeLinearOperator((proj, res_model))
x_fwd2 = proj_with_res_model(f_gpu)
# Incorporating image-based resolution model and adding the effect of attenuation
x_att_fwd = proj(y_gpu)  # "y_gpu" is the attenuation image
att_sino = xp.exp(-x_att_fwd)
att_op = parallelproj.ElementwiseMultiplicationOperator(att_sino)
proj_with_att_and_res_model  = parallelproj.CompositeLinearOperator((att_op, proj, res_model))
x_fwd3 = proj_with_att_and_res_model(f_gpu)
# Defining the ramp filter
phi_max = xp.pi
num_rad = 111
num_phi = 210
r = xp.linspace(-55.5, 55.5, num_rad, device=dev, dtype=xp.float32)
n_filter = r.shape[0]
r_shift = torch.arange(n_filter, device=dev, dtype=torch.float64) - n_filter // 2
f = torch.zeros(n_filter, device=dev, dtype=torch.float64)
f[r_shift != 0] = -1 / (torch.pi**2 * r_shift[r_shift != 0] ** 2)
f[(r_shift % 2) == 0] = 0
f[r_shift == 0] = 0.25

# FBP of the created sinogram using the complete projector
pre_corrected_sino3 = x_fwd3[:, :, 17]
filtered_pre_corrected_sino3 = 1.0 * pre_corrected_sino3

for i in range(num_phi):
    filtered_pre_corrected_sino3[:, i] = xp.asarray(
        np.convolve(
            parallelproj.to_numpy_array(filtered_pre_corrected_sino3[:, i]),
            f.cpu(),
            mode="same",
        ),
        device=dev,
    )
FBP_x_fwd3= proj_with_att_and_res_model.adjoint(filtered_pre_corrected_sino3.unsqueeze(2).repeat(1, 1, 35))

[image]

gschramm, 2024-10-23:
Hi,

1. For FBP you need to "pre-correct" for attenuation (divide by the multiplicative attenuation factors).
When simulating data without noise there should be no difference the case (i) no attenuation in the data simulation and no precorrection and (ii) attenuation simulation followed by precorrection before FBP.
2. Which projector are you using? FBP needs "parallel view projections" (use ParallelViewProjectors). The more realistic PET detector based projectors do not use strict parallel views because the do zig-zag sampling.
3. To do a standard FBP, I would use a projector that does not include an image based resolution model. If you do, your FBP gets post-filtered siche the adjoint of the projector with resolution model is "backprojector and then gaussin filter in image space".
4. The projector used for data generation is in general independent from the one used for reconstruction. The first one should simulated the PET physics as realistic as possible (e.g. resolution effects, attenuation, ...). The 2nd one can be a simplication.

hamidrkKTH, 2024-10-23:
Thanks for your reply.
1) I use your provided simulated brainweb PET/MR data sets. For this dataset which value should I use to "pre-correct" for attenuation?
2) I am using the following geometry. I set max_ring_difference=0, so I do not have oblique LOR.
num_rings = 35
scanner = parallelproj.RegularPolygonPETScannerGeometry(
    xp,
    dev,
    radius= 211/2 - 2,
    num_sides=12,
    num_lor_endpoints_per_side= 35,
    lor_spacing= 211*np.pi/(35*12),
    ring_positions=xp.linspace(-20, 20, num_rings),
    #ring_positions=torch.linspace(-25, 25, num_rings),
    symmetry_axis=1,
)

lor_desc = parallelproj.RegularPolygonPETLORDescriptor(
    scanner,
    radial_trim=155,
    max_ring_difference=0,
    sinogram_order=parallelproj.SinogramSpatialAxisOrder.RVP,
)
proj = parallelproj.RegularPolygonPETProjector(
    lor_desc, img_shape=(220, num_rings, 220), voxel_size=(80/220, 40/num_rings, 80/220)
3) For FBP, I was using the same geometry as I used to generate the sinograms.

gschramm, 2024-10-24:
1. FBP assumes that the acq. data comes from lines integrals. In PET they come from attenuation + normalization weighted line integrals (we also usually simulated the effect of attenuation). That means if you include the effect of attn. in your data simulator, you have to pre-correct (divide the data by the attenuation sinogram) before calling FBP.

2. FBP assume that view have acquired parallel views. which is not the case in real PET scanners.
To simulate data, it is better to use a "ParallelViewProjector" 
https://parallelproj.readthedocs.io/en/stable/python_api.html#parallelproj.projectors.ParallelViewProjector3D
Note, however, that the data from real PET scanners usually includes zig-zag sampling (not perfectly parallel views), as shown here:
https://parallelproj.readthedocs.io/en/stable/auto_examples/01_pet_geometry/02_run_regular_polygon_pet_sino.html#sphx-glr-auto-examples-01-pet-geometry-02-run-regular-polygon-pet-sino-py

3. Using the same projector for simulation + recon is ok, but is called the "inverse crime". In real life the projector we use for recon is always a "simplied model" of the real acq. physics.

hamidrkKTH, 2024-10-27:
Thanks for the answers and the explanations.

1) I got it.
Now, for FBP, I use 2D "ParallelViewProjector". However, when I use it based on the sinogram size, the size of reconstructed image is different from the original image, and when I change the parameters of the projector to get the same size image, the quality of the image becomes corrupted.

gschramm, 2024-10-27:
You have to make sure that the "radial / angular" coordinates of the view in the projectors match. Note that this is not trivial and probably requires resampling.

closed 2024-10-27 by gschramm