KUL-recon-lab/parallelproj #103: Behavior of non-TOF backprojection

lhell1, 2024-12-13:
Hello! 

I am experiencing an issue when back-projecting a sinogram full of ones. I chose 360 evenly sampled LORs in a circle around the object (like a CT scan). After applying the bwd_layer, the volume does not exhibit a maximum value of 360 - as I would have expected - but severely higher values (~factor 15). Is this expected behavior? Am I missing a factor or something else related to Joseph's method/geometry? I tried to look for an answer in the referenced paper but could not find it.
 
Thank you in advance!

gschramm, 2024-12-13:
Parallelproj's Joseph implementation calculates the line integrals trough a 3D image respectiving the voxel size (length of the ray) in the forward projector.
And the backprojector is the exact adjoint of that.

lhell1, 2024-12-13:
This does not seem to make sense to me... Assume we have a voxel in our image that is hitted exactly in its center by each LOR. This would mean the entries in the system matrix (bilinear interpolaton) are all exactly one. Now assume that all projection values that cross this voxel are one. The adjoint of the system matrix (= transpose) should add 360 times a one to the image voxel. This should result in a maximum value of 360 and not more. Could you elaborate your statement concerning this example?

gschramm, 2024-12-13:
0. To verify whether things make sense I suggest the following: (a) create am image full of zeros and single voxel containg a one. (b) Forward project that image and verify if the calculated line integrals are correct (they should be according to the units tests). (c) verify that the back projector is the exact adjoint of the forward projector (should be according to the unit tests).
1. The value that you will get in the backprojection will depend on the voxel size of your image.
2. How do you generate the sinogram coordinates from the detector coordinates? Are you doing that yourself or resusing the PET sinogram classes? If you do that you have to be carefule since they contain also oblique planes which add to the back projection.

lhell1, 2024-12-16:
I am defining my projector lines myself with ListmodePETProjector. Not sure what you mean with "sinogram coordinates from the detector coordinates"? Should they not be the same? I will have a look at the unit tests and adjointness. Thanks!

gschramm, 2024-12-16:
If you want a 2D parallel view projector that resembles parallelprojections at different angles, have a look at 
https://parallelproj.readthedocs.io/en/stable/python_api.html#parallelproj.projectors.ParallelViewProjector2D
such that we can rule out that there are errors in the LOR start/endpoint definition.

Do you see the same behavior there?

lhell1, 2024-12-17:
Hello again, I think I figured out where my confusion came from. For the parallel geometry everything seems to be the way I stated above. The behavior I was confused about was in an cone-beam-like geometry, but it does make sense now and parallelproj seems to be exact. I am now figuring out what the framework I usually work with (ITK RTK) is doing differently for Joseph forward and backward projection. They seem to use some kind of weighting procedure that counteracts geometry differences. Which is why the projection of that framework and parallelproj did not seem to align. Thank you a lot for your help in trying to understand!

closed 2024-12-17 by lhell1

gschramm, 2024-12-17:
Glad I could help. Let me know know if you figured out the difference to RTK.
It could be that different (non-Joseph) projectors are used in RTK. The easiest is to compare forward and back projections of image/sinograms containing a 1 in a single pixel and 0s everywhere else.