KUL-recon-lab/parallelproj #109: Issue Regarding import parallelproj error

zjutk, 2025-01-16:
Dear developer,

I hope this email finds you well. I encountered an error when attempting to run the open-source code you provided on my Windows system, which has left me somewhat puzzled:

> AttributeError
> function 'get_cuda_device_count' not found
>   File "D:\ltk\parallelproj-1.10.1\examples\01_pet_geometry\01_run_regular_polygon_pet_scanner.py", line 23, in <module>
>     import parallelproj
> AttributeError: function 'get_cuda_device_count' not found

 I had previously run this code successfully on a Linux system without encountering this issue. After searching online for potential solutions, I was unable to find one that fits my situation, so I am reaching out to you for assistance.

I suspect that the issue may be related to an outdated version of the parallelproj_cuda.dll file (though I am not entirely sure). Could you kindly provide a link to a newer version of parallelproj_cuda.dll?

Additionally, I have attached the list of installed packages and dependencies in the virtual environment I created for your reference.

[image]
[image]

I look forward to your response and assistance. Thank you for your time and support.

Best regards,
Kun Tian

gschramm, 2025-01-16:
Many thanks for reporting this issue

Which cuda version are you running?
Can you post the output of:
```
nvidia-smi -l
```

zjutk, 2025-01-16:
Sure! It is my cuda version information:

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2020 NVIDIA Corporation
Built on Mon_Nov_30_19:15:10_Pacific_Standard_Time_2020
Cuda compilation tools, release 11.2, V11.2.67
Build cuda_11.2.r11.2/compiler.29373293_0

[image]

gschramm, 2025-01-16:
Interesting, `nvidia-smi` you are running cuda 12.6, but your comment says 11.2

Can you test the following (creating two test envs):

```
conda create -c conda-forge -n test-cuda11 cuda-version=11 parallelproj
```
and
```
conda create -c conda-forge -n test-cuda12 cuda-version=12 parallelproj
```
and see if the error persists?

zjutk, 2025-01-16:
Yes, I observed the same error in two different virtual environments: 

AttributeError: function 'get_cuda_device_count' not found.

gschramm, 2025-01-16:
Hm. I just tried with my local windows running cuda 12.4 and the examples / package import runs without problems.
The `get_cuda_device_count()` function is indeed part of `libparallelproj_cuda`.

Can you execute the following:

```python
import os
from ctypes.util import find_library

lib_parallelproj_cuda_fname = find_library("parallelproj_cuda")

print(ib_parallelproj_cuda_fname)
print("-----")
print(os.environ["PARALLELPROJ_CUDA_LIB"])
```

zjutk, 2025-01-17:
Thank you for your response. After running the instructions you provided, I obtained the following results:

[image]

gschramm, 2025-01-17:
Ok. That explains it. On your system you have the environment variable `PARALLELPROJ_CUDA_LIB` which points to `D:\parallelproj.dll` (probably an older version of the lib that you created manually).

If you unset the env. variable, the correct lib in your `anaconda_env\test_cuda12` will be autonatically loaded.

You should only use the `PARALLELPROJ_CUDA_LIB` env. variable, if the auto detection of the lib fails.
The welcome screen after `import parallelproj` will show you which lib is being used.

closed 2025-01-17 by gschramm